ZhangAoCanada/RADDet #4: Auto-Annotation Code

chiyukunpeng, 2021-07-28:
Hi, it is a good work! 
I am very interested in auto-annotation method. Can you provide this code? 
Thanks!

ZhangAoCanada, 2021-07-28:
Hi,

Thanks for your interest.

The reason I didn't open-source the source code for the auto-annotation is because the work was done with Sensor Cortek Inc. So technically, they have the right to decide whether we open-source it. 
In addition, I only kept the original development repo. The scripts inside are unorganized, which may make the repo itself a little bit hard to read. :\ 
I can try to ask their permission for the original development repo if you are still interested.

chiyukunpeng, 2021-07-29:
Thanks for your reply!

closed 2021-07-29 by chiyukunpeng