ZhangAoCanada/RADDet #5: ValueError: Cannot read sequences.txt.

vinayak2914, 2021-11-08:
<ipython-input-19-eca9ae57661d> in main()
     50     ### NOTE: preparing data ###
     51     print ('###########################################')
---> 52     data_generator = DataGenerator(config_data, config_train, config_model,model.features_shape, anchor_boxes,anchors_cart=None, cart_shape=None)
     53     train_generator = data_generator.trainGenerator()
     54     validate_generator = data_generator.validateGenerator()

/content/drive/My Drive/Colab Notebooks/RADDet/dataset/batch_data_generator.py in __init__(self, config_data, config_train, config_model, headoutput_shape, anchors, anchors_cart, cart_shape)
     25         self.anchor_boxes = anchors
     26         self.anchor_boxes_cart = anchors_cart
---> 27         print ('###########################################')
     28         self.RAD_sequences_train = self.readSequences(mode="train")
     29         self.RAD_sequences_test = self.readSequences(mode="test")

/content/drive/My Drive/Colab Notebooks/RADDet/dataset/batch_data_generator.py in readSequences(self, mode)
     64                                 "RAD/*/*.npy"))
     65              print (sequences)
---> 66         else:
     67             sequences = glob.glob(os.path.join(self.config_data["test_set_dir"], \
     68                                 "RAD/*/*.npy"))

ValueError: Cannot read sequences.txt.                         Please check if the file is organized properly.

vinayak2914, 2021-11-08:
I am not able to find the sequences.txt in the github repository

xuzekai1997, 2021-12-02:
I have the same problem too.

ZhangAoCanada, 2021-12-09:
Hi, yall. Thanks for reporting the issue and apologize for the late reply.

I think the issue is caused by wrong "train_set_dir" or "test_set_dir" in `config.json`,
Or the dataset is organized in the wrong format. (See `README.md` for the correct format).

About the `ValueError` you posted, it was originated from my working repo and I forgot to change it. I will fix this soon.

Let me know if there is anything else.

ZhangAoCanada, 2021-12-23:
98b20d623ea8cf8874e0c1227316c550a32b50ea : The err msg has been fixed.

closed 2021-12-23 by ZhangAoCanada