ZhangAoCanada/RADDet #15: How to read raw adc data?

Mirage1089, 2022-03-07:
Hello,

I am interested to use your raw adc data that you provided (https://drive.google.com/drive/folders/1G5Vfzioph46mjeHbz4OXJb2yHeSGS8cd). For that I need to read the data. Do you have any script to read it? It seems every file represents 1 frame.  And In the file it is written, the shape is ```'shape': (256, 64, 4, 2)```, Which means 256 sampling points and 64 chirps. What 4 and 2 means? Also want to know how many azimuth antennas you used.

Thank you.

ZhangAoCanada, 2022-03-08:
Hi, 

Are you talking about FFT (for the raw data)? If so, I used the FFT tools developed by SensorCortek during my internship there. So there is no relative script. But there is a script called `visualize_testdata.py` to guide you how to visualize the data after FFT.

For the other question, 4 and 2 stand for 4 on-board receivers and 2 transmitters (AWR1843boost).

Thanks

Mirage1089, 2022-03-08:
Hello,

The raw adc data means the data you received from your radar (AWR1843boost) sensor. This is not the result of fft data before the fft. You already mention that we can download the raw adc data and you recorded 10158 frames. these data are availabe at (https://drive.google.com/drive/folders/1G5Vfzioph46mjeHbz4OXJb2yHeSGS8cd) where I believe each file contains a frame.

You mentioned it here,
[image]

I am not sure how did you record the data. If you use ```DCA1000 board``` and mmwave studio then it will record in a ```.bin \ binary file```. Here in the adc folder you have ```.npy file``` where each file contains 1 frame and it seems it saves also binary data.

These ```.npy file \ each frame data``` is needed to read and then do some signal processing (that you already mentioned in your paper) then you can have the image to apply the ML\YOLO.

If I could not explain myself or have some misunderstanding about you data please tell me.

Thank you.

ZhangAoCanada, 2022-03-09:
yes, you are absolutely right.

Perhaps I didn't explain myself well last time.

So what I meant was: the entire process of how to use `AWR1843boost` and `DCA 1000` to capture raw radar data is developed by SensorCortek Inc., which is a Canadian company. I was working there as an intern and was responsible for processing those raw radar data. I don't know much about how to capture the raw data.
They also built a tool for processing radar signals and save them as `.npy`. The data that I shared with you were derived from that tool.

So, if you are asking me how to get those `.npy` data, I would suggest you to contact SensorCoretek Inc. directly. [Their official website](https://sensorcortek.ai/) and [Their contact page](https://sensorcortek.ai/team/). You can email their CEO or COO if you want.

If you are asking how to do FFT on those `(256, 64, 4, 2)` matrices, I can help you with that.

Let me know if there is any further questions.

Thanks,

closed 2022-09-29 by Mirage1089