ZhangAoCanada/RADDet #17: Question regarding Angular Resolution and Angular Range

syed-mujtaba-hassan, 2022-04-05:
Hi Zhang,

Thanks for your great work and sharing your dataset. 

I have a question regarding the dataset. Will it be possible to specify what is the range of angular values. From your plot, it appears that the range is around -90 to 90 degree. However, in the readme file, you specified the angular resolution as 0.006135923 degrees/bin. I am not sure I understand this resolution because 0.006135923*256=1.5 degrees which is much less than 180 degree. Also, can you tell whether the bins represent uniform angles. I have read that the fft and angle do not have a linear relationship but a sin relation. In that case, will an assumption for a linear relationship between angle and fft bins works, especially at higher angles. 

Thanks a lot for the help.

Best Regards,
Mujtaba

ZhangAoCanada, 2022-04-06:
Hi,

Thanks for your questions.

1. Thanks for pointing out the problem, the angular resolution should be **0.006135923 radian/bin**.  I will try to fix the README when I have my laptop back.
2. Yes, each bin stands for a constant range of angles. You are right about non-linear relation (sine) in FFT. I was told by a TI tech guy: because of the special raw data encoding system, using sine function will improve the accuracy of FFT outputs. Meanwhile, when reading the outputs, the bin size should be kept linear.
3. Still about question 2: you can also try other radars to see if they are the same (probably not :\).

Let me know if there is anything else.

Best,
Ao

syed-mujtaba-hassan, 2022-04-06:
Hi Ao,

Thanks for your reply. Helps a lot.

I have a question. What is the maximum range of angular values. If resolution is 0.006135923 rad/bin, then the angle range is   0.006135923 * 180 / pi *256 = 90 degree. So, is the angular range going from -90 to 90 degree or from -45 to 45 degree ?

Thanks a lot. 

Best Regards,
Mujtaba

ZhangAoCanada, 2022-04-07:
Your question got me there for like 10min......

It came to me that: the reason why I typed 0.006135923 degrees/bin in the first place is that I directly copied the configuration setups from one of my friends when I was working at SensorCortek. Btw, the entire radar capture library for this project was developed during my internship at SensorCortek. 

The exact angular range should be around -86 degrees to 86 degrees, according to my computation. As for the angular resolution, I may have to contact my friends for the exact numbers.

Thanks,
Ao

syed-mujtaba-hassan, 2022-04-07:
Thanks a lot for the reply. Please tell about the angular resolution once you found it. 

Best Regards,
Mujtaba

syed-mujtaba-hassan, 2022-04-13:
Hi Ao,

I have found that you are using a formula to convert RA values into Cartesian coordinates inside "toCartesianMask" function. In this formula, you are using a sine relationship between angle and bin and not a linear one . However, in your visualization function "imgPlot", you are using linear relationship between angle and bin to plot RA map. Can you tell me why this is so ?

Thanks a lot.

Mujtaba

Best Regards,
Mujtaba

ZhangAoCanada, 2022-04-14:
Hi,

The angle decoding format is from Sensor Cortek. Also, I was told that this format is originated from TI. They also told me apart from this decoding format, the output bins could be read as linear. That's the reason I did it.

If you have further questions about how it works, I suggest you email Sensor Cortek or TI to get the answers :\.

Thanks,
Ao