ZhangAoCanada/RADDet #22: About running the project

siddupp, 2022-10-15:
Hi sir 
Can you please help me to run this project i didn't understand what should I run first and what next.
          So can you please provide me the  "sequence of the python files or the commands to be executed "to get the output.

ZhangAoCanada, 2022-11-25:
First, you have to install all the required pip packages, simply run
```bash
pip3 install -r requirements.txt
```

Next, you need to follow the `README.md` to prepare dataset.

With dataset being ready, you can run do the following things:
- for training 3d radar Object Detection.
  ```python
  python3 train.py
  ```
- for training 3d radar Object Detection.
  ```python
  python3 train_cart.py
  ```
- for testing:
  ```python
  python3 evaluate.py
  ```
- for visualization of the dataset itself
  ```python
  python3 visualize_testdata.py
  ```

In addition, you can change your GPU id with the code below:
```python
import os
os.environ["CUDA_DEVICE_ORDER"]="PCI_BUS_ID"
os.environ["CUDA_VISIBLE_DEVICES"]="0"
```
change the value `0` to the GPU id you want.

Hope I made this clear.

siddupp, 2022-11-25:
> First, you have to install all the required pip packages, simply run
> 
> ```shell
> pip3 install -r requirements.txt
> ```
> 
> Next, you need to follow the `README.md` to prepare dataset.
> 
> With dataset being ready, you can run do the following things:
> 
> * for training 3d radar Object Detection.
>   ```python
>   python3 train.py
>   ```
> * for training 3d radar Object Detection.
>   ```python
>   python3 train_cart.py
>   ```
> * for testing:
>   ```python
>   python3 evaluate.py
>   ```
> * for visualization of the dataset itself
>   ```python
>   python3 visualize_testdata.py
>   ```
> 
> In addition, you can change your GPU id with the code below:
> 
> ```python
> import os
> os.environ["CUDA_DEVICE_ORDER"]="PCI_BUS_ID"
> os.environ["CUDA_VISIBLE_DEVICES"]="0"
> ```
> 
> change the value `0` to the GPU id you want.
> 
> Hope I made this clear.

Thank you for the reply sir😊
       While running the visualization of testdata.py file i am getting the error in this line "config = loader.readConfig()" loader has no attribute named readconfig can you please suggest me the solution?

ZhangAoCanada, 2022-11-25:
You may need to goto the project root dir to run this script. It wull call the `readConfig()` function in `util/loader.py`.

siddupp, 2022-11-25:
Thank you for your response
I have tried to call the function from the root directory only but i am
getting error

siddupp, 2022-11-25:
Can you please share the trained waits so can i use it directly for the
testing purpose

ZhangAoCanada, 2022-11-28:
Hi, the pre-trained weights are already provided in `README.md`.

I am attaching it here for convenience. https://drive.google.com/drive/folders/1EZQtcAUzVKYmYHDuiiV_FB5Obk7PbSoE?usp=sharing

siddupp, 2022-11-28:
Thanks sir , I'll check it out.

Gabo181, 2024-01-25:
Hey, in evaluate.py you're loading the data for the plot.
[image]
In line 219, 233 and 238 the code is trying to access paths "input_dir, image_dir and gt_dir which arent in the config. Therefore an error occurs with "file not found".

are these just the test directorys and then the gt, rad and image directorys?

thanks!