ZhangAoCanada/RADDet #25: In the data label, the details of the combination of image and radar

ciwei123, 2022-11-25:
@ZhangAoCanada Hi，Thanks for your sharing. I don't know if my understanding is correct about figue2 in your paper. 
1. image branch:  Get the depth map and instance segmentation through the image, and then get the 3D box , and then map this 3D box to the RA map, and final get the 2D box in RA map.
2. radar branch:  We can obtain some possible target radar points through the clustering method.

How to integrate the **2D box** with **the radar cluster points** in RA map, for example, for some 2D boxes, there is no corresponding radar cluster points. I am very much looking forward to your reply, thanks again。

ZhangAoCanada, 2022-11-25:
Hi, thanks for your message.

For camera: after depth map and segmentation, I get 3D points, not 3D boxes. Then those 3D points are projected to 2d points. Based on the segmentation, we can cluster these 2D points.

For radar: the obtained radar points are then processed with DBSCAN to generate clusters.

The matching process is a GNN-like algorithm (think it as Hungarian Algorithm). This means, when there is a radar cluster, we match it with the globally closest camera cluster. 
If there are radar clusters not being labelled, we will manually fix them after the process.
If there are camera clusters not being matched, we will try different radar pre-process methods. If the problem stays, we will delete this frame.

That's how we did the annotations........

ciwei123, 2022-11-25:
@ZhangAoCanada Thank you very much for your quick reply！ If it is considered as Hungarian Algorithm，how to calculate the similarity between two clustered point? Further, could you tell me the specific matching algorithm, such as which code or paper I can refer. Thank you very much!

ciwei123, 2022-11-25:
@ZhangAoCanada Another question, is there lidar? Why not use lidar to make labels?  I am very much looking forward to your reply, thanks again!!

ZhangAoCanada, 2022-11-25:
Hi,

I used a very simple method, Euclidean distance between the center points of two clusters. The matching algorithm i used was Hungarian Algorithm. You can try better matching algorithms to reach higher annotation precision....

The reason why not using Lidar: we purchased a Lidar at the time, but the delievery was delayed due to COVID-19. I had to figure out another way to annotate radar data (stereo cameras)....

ciwei123, 2022-11-25:
@ZhangAoCanada Thank you very much for your answer, I understand, Thanks again.

closed 2022-12-12 by ZhangAoCanada