ZhangAoCanada/RADDet #27: intrinsics matrix and extrinsics  matrix

anxiaohuai, 2022-12-18:
Hello, do you have the intrinsics matrix and extrinsics  matrix after camera calibration? How can I get it?

ZhangAoCanada, 2022-12-19:
Hi, I didn't upload those matrices. The camera calibration was done with SensorCortek. They have all copy rights.

You may have to ask for their permissions if must.

anxiaohuai, 2022-12-22:
Could you please tell me how to contact SensorCortek? Because I think only by obtaining these parameters can I obtain complete and extensible data

ZhangAoCanada, 2022-12-29:
[calibration_results.zip](https://github.com/ZhangAoCanada/RADDet/files/10317354/calibration_results.zip)
I have contacted them myself. The stereo calibration results are attached.
All results were acquired from OpenCV.

I hope they help.

closed 2023-03-06 by ZhangAoCanada