ZhangAoCanada/RADDet #28: Range-Azimuth-chrips

Acciwu, 2022-12-23:
Hi! I would like to do only two FFTs on your ADC data to get range-azimuth-chrips. I used the OpenRadar SDK, first do the FFT on the range, then on the azimuth, can I get range-azimuth-chrips? Are other radar parameters required? look forward to your reply!

Acciwu, 2022-12-23:
[image]

Acciwu, 2022-12-24:
Is it okay to perform distance FFT first and then directly perform azimuth FFT? Can you help me, thank you very much!

import numpy as np
import matplotlib.pyplot as plt

c=3.0e8;  
B=768e6;       # FM bandwidth
K=30e12;       # chirp rate
T=B/K;         # FM period
Tc=160e-6;     # Chirp total cycle
fs=10e6;       # sampling rate
f0=77e9;       # Initial frequency
Lambda = c / f0;   # Radar signal wavelength
d=Lambda / 2    # Antenna array spacing
n_samples=256; # Number of sampling points/pulse
N=256;         # FFT points in range direction
n_chirps=64;  # Number of pulses per frame
M=64;         # Doppler FFT points
n_RX=8;        # Number of RX antenna channels
# Q = 180      # angle FFT points
Q = 256;       # angle FFT points
xx = 1;        # Frame xx

data_radar = adc_data
# range FFT
range_win = choose_windows('Hamming', n_samples)   # add Hamming Windowing
doppler_win = choose_windows('Hamming', n_chirps)
range_profile = np.zeros((256, 64, 8))
for k in range(n_RX):
    for m in range(n_chirps):
        temp = data_radar[:,m,k] * range_win    # Hamming Windowing
        temp_fft=np.fft.fft(temp, N, 0)    # Do N-point FFT for each chirp
        range_profile[:,m,k] = temp_fft
print(range_profile.shape)

# azimuth FFT
angle_profile = np.zeros((256, 64, 256))
for n in range(N):  # range
    for m in range(M): # chirps
        temp= range_profile [n,m,:]
        temp_fft=np.fft.fftshift(np.fft.fft(temp, Q))
        angle_profile[n,m,:] = temp_fft
print(angle_profile.shape)

# in 0 chrip
plt.imshow(abs(angle_profile[:,0,:]))
[image]

ZhangAoCanada, 2022-12-29:
Hi, thanks for the question.

To answer your question: Yes, you can directly perform DOA after RangeFFT.

To explain why: the way of data formation is as follow. Each radar frame contains 1 single chirp per receiver. That means, you can collect 8 chirps from 2 transmitters * 4 receivers. The Doppler axis is to accumulate several (64 in my configurations) consecutive radar frames and then perform FFT on that axis.
If you are using my ADC data (256, 64, 8), you can perform RangeFFT on 256 and AzimuthFFT on 8. Then sum up the 64 and normalize them. That will do.

One thing to NOTE: do remember to do FFTshift after FFT, if you want to get readable results.

Thanks,

braundeniz, 2023-06-23:
So am i right, that i cant use the RAW ADC data from your dataset directly? 

Do i have to perform the FFT's, proposed from @Acciwu first?

Thanks in advance

braundeniz, 2023-06-23:
Because i loaded the RAW Data downloaded from your Drive and they have the shape [256,256,64], which should be the shape for the RAD Tensor am i right?

So in my understanding the Data from the OneDrive is already in RAD Format?

Gabo181, 2024-02-16:
Hey @braundeniz , did you get any results to bring your data cube into (256, 256, 64)

huiwenXie, 2024-06-21:
@Acciwu Hello, I have some question to ask you. I found that the data size is 256*64*8, do you know that the data is " 1 frame with 64 chirp“  or "64 frame and each frame has 1chirp"? Additionlly, if the data is  " 1 frame with 64 chirp“, do you know the  time inteval between each chirp?  If the data is  " 64 frame and each frame has 1chirp“, do you know the  time inteval between each frame? Thanks a lot! If you are willing to answer  my quesiton, I will be very happy.