ZhangAoCanada/RADDet #29: Issue while downloading data from google drive

javalisiddu, 2023-02-01:
Hi, I am unable to download the dataset from the link you have provided in your repository. I even tried requesting access. But its not happening. Please address this issue.

ZhangAoCanada, 2023-02-03:
Hi, 

You may have to use [oneDrive](https://uottawa-my.sharepoint.com/personal/azhan085_uottawa_ca/_layouts/15/onedrive.aspx?id=%2Fpersonal%2Fazhan085%5Fuottawa%5Fca%2FDocuments%2FRADDet%5FDATASET&ga=1) for data downloading now.
My google drive account has been deleted by my university, as it has been 2 years since my graduation.
For additional data access (ADC and etc), I don't have the backup of them. I may need to contact my university......

monroyaume5, 2023-02-13:
> 

Hi, I encountered the same problem and I really need ADC data for my own research. I really appreciate you sharing this dataset, your dataset is the best I have found so far, if ADC data is also available it will make this dataset perfect. I would be very grateful if you could manage to contact your university to resolve this issue!

ZhangAoCanada, 2023-03-06:
Thanks y'all for the interests. 

I have tried contacting my university. They told me they have stopped the cooperation with Google Drive and switched to One Drive. The data could not be migrated between these two. So, they had to delete all university google drive accounts...

erics24, 2023-09-21:
Has anyone been able to successfully unzip RADDet_DATASET.zip that we download from OneDrive?  I've had quite a few problems with headers errors and others preventing successful extraction.  Thanks.

YDMYy, 2023-10-19:
> Has anyone been able to successfully unzip RADDet_DATASET.zip that we download from OneDrive? I've had quite a few problems with headers errors and others preventing successful extraction. Thanks.

I also encountered this problem, so I skipped these directly....

lijiunderstand, 2024-01-04:
> Thanks y'all for the interests.
> 
> I have tried contacting my university. They told me they have stopped the cooperation with Google Drive and switched to One Drive. The data could not be migrated between these two. So, they had to delete all university google drive accounts...

Hi, Can you update the ADC data link and put the data from link https://drive.google.com/file/d/1R3dalVn9_x8GIH7lm3lZIUl1ftOzoJnp/view to your oneDrive in case of link is unavailable again? I think this dataset is very important for radar researchers, Thanks for your help.

ZhangAoCanada, 2024-01-05:
> > Thanks y'all for the interests.
> > I have tried contacting my university. They told me they have stopped the cooperation with Google Drive and switched to One Drive. The data could not be migrated between these two. So, they had to delete all university google drive accounts...
> 
> Hi, Can you update the ADC data link and put the data from link https://drive.google.com/file/d/1R3dalVn9_x8GIH7lm3lZIUl1ftOzoJnp/view to your oneDrive in case of link is unavailable again? The data is provided by [Acciwu](https://github.com/Acciwu), I think this dataset is very important for radar researchers, Thanks for your help.

Thank you for your share.
I tried to download the ADC data and upload them to oneDrive, but only found out that I have no more access to edit the University oneDrive. They probably have locked down my account since I have graduated for a while...

lijiunderstand, 2024-01-05:
> Thank you for your share. I tried to download the ADC data and upload them to oneDrive, but only found out that I have no more access to edit the University oneDrive. They probably have locked down [...]

Got it, I'd like to help, I also upload the adc data to my baidu netdisk, the link is https://pan.baidu.com/s/1T3p5wrxgy0gdsZBRFqapVQ?pwd=szax, and the time limit I set is permanent. Can you help to update the adc data link with this two links in your readme? Thank you.