ZhangAoCanada/RADDet #34: How to match angle bins with the true angles

monroyaume5, 2023-03-31:
I am just wondering how you transform the angle bins into real angle values. In the code, I have found that the 

> "angular_resolution" : 0.006135923

Based on this angular_resolution, I can't get the angle values as those presented in the **drawer.py** as shown below:

[image]

So could you please share with me how do you transform the angle bins [0, 64, 128, 192, 255] into real angles [-85.87, -42.93, 0, 42.93, 85.87]?

ZhangAoCanada, 2023-04-04:
The angular resolution is not accurate in `README.md`

The code for transformation from doppler bins to Cartesian was developed by Sensor Cortek. For detailed info, you may have to ask them.....
But I can give you a hint of how to do so.
```python
    for i in range(RA_mask.shape[0]):
        for j in range(1, RA_mask.shape[1]):
            if RA_mask[i, j] > 0:
                point_range = ((radar_config["range_size"]-1) - i) * \
                                radar_config["range_resolution"]
                point_angle = (j * (2*np.pi/radar_config["azimuth_size"]) - np.pi) / \
                                (2*np.pi*0.5*radar_config["config_frequency"]/ \
                                radar_config["designed_frequency"])
                point_angle_current = np.arcsin(point_angle)
                if point_angle_previous is None:
                    point_angle_previous = point_angle_current
                for point_angle in np.linspace(point_angle_previous, point_angle_current, \
                                                gapfill_interval_num):
                    point_zx = polarToCartesian(point_range, point_angle)
                    new_i = int(output_mask.shape[0] - \
                            np.round(point_zx[0]/radar_config["range_resolution"])-1)
                    new_j = int(np.round((point_zx[1]+50)/radar_config["range_resolution"])-1)
                    output_mask[new_i,new_j] = RA_mask[i, j] 
                point_angle_previous = point_angle_current
```

monroyaume5, 2023-04-05:
Thanks for the sharing. Another question, I'm been using your data to train the model and get an ideal result by using your data to do the test. However, when I use the model to perform the test by using my own data, it just gave me a very bad result. I use the same TI1843 radar board but with a different configuration(i.e, different range resolution and such thing), so do you have any idea about this phenomenon? Are the distributions of radar data received by different radars (despite the same model, both TI1843) different? Or does the difference in configuration make a big difference?

ZhangAoCanada, 2023-04-12:
I do believe the reason goes to different configurations. E.g., different maximum detection range and doppler bins will give you totally different radar spectrum. As your model has probably overfitted on my dataset, it may not be able to recognize the features from your data.
I am suggesting combining both my data and your data, then train the model again (from ckpt if you want).

That's my best guess.