ZhangAoCanada/RADDet #35: The link for raddet data on the google drive link is not available.

yujheli, 2023-04-08:
Hi, I would like to request the link for the data sharing of the ADC signal. The link on google drive is not available now and I can not find the ADC folder on the ONE drive. Thanks!

ZhangAoCanada, 2023-04-12:
I am aware of the unavailable google drive link, since the google drive account was deleted by my university. What's more is that, I don't have any backup ADC data locally because they take tooooooo much space to store :(.

Sorry for the inconvenience.

colindecourt, 2023-06-23:
Hi, 
I found a file on my computer with the ADC data downloaded from Google Drive. It's around 2.2GB. I could share it with you so that you can create a new Google Drive folder.

YDMYy, 2023-10-02:
> Hi, I found a file on my computer with the ADC data downloaded from Google Drive. It's around 2.2GB. I could share it with you so that you can create a new Google Drive folder.

Hello, I also need this document. Can you share it with me? Thank you very much.

ZhangAoCanada, 2023-10-07:
> Hi, I found a file on my computer with the ADC data downloaded from Google Drive. It's around 2.2GB. I could share it with you so that you can create a new Google Drive folder.

That would be perfect! Please let me know where the share link is if you finish uploading.
Thank you in advance, you are totally a life-saver.

lijiunderstand, 2023-12-19:
> Hi, I found a file on my computer with the ADC data downloaded from Google Drive. It's around 2.2GB. I could share it with you so that you can create a new Google Drive folder.

Hi, Could you share the ADC data? Thank you

goyalsuraj, 2024-06-15:
> Hi, I found a file on my computer with the ADC data downloaded from Google Drive. It's around 2.2GB. I could share it with you so that you can create a new Google Drive folder.

Hello, do you still have the ADC data available?

taj1290, 2024-08-08:
https://drive.google.com/file/d/1R3dalVn9_x8GIH7lm3lZIUl1ftOzoJnp/view

ADC data.

BJ-LhR, 2025-03-18:
Hello! I noticed that the download link for the raw ADC dataset doesn't have the directories 'gt_box/' and 'gt_box_test/'.
Could you please provide the downloads for these two files? Or could you tell me how to generate these two folders?
They are of great significance for my work, especially since they hold the vital ground - truth annotations required for validation and testing.
Your help in this matter would be really appreciated.