ZhangAoCanada/RADDet #40: Help about converting ADC data to RAD data

cuixinyuan70, 2023-06-28:
Thank you for your brilliant work first! And I am so sorry to bother you. I transform the raw ADC data with a (sample, chirps, Tx, Rx) structure to (sample, chirps, vritualRx), which is (256, 64, 8).  Then do zero-padding before Azimuth FFT, making the ADC data to (256, 64, 256). Finally do 1D-FFT at the sample latitude to obtain the range, then 2D-FFT at the chirps latitude to obtain the Doppler, and 3D-FFT on the Rx latitude to obtain the angle.

The problem is that I cannot match the RA&RD heatmap generated by myself with the RAD data RADDet provided. figure_1 is generated by me, figure_2 is provided by RADDet RAD.

cuixinyuan70, 2023-06-28:
[image]
[image]

cuixinyuan70, 2023-06-28:
Here are my codes:
# zero-padding
data_radar = np.zeros((256, 64, 256)).astype(complex)
for k in range(256):
    for m in range(n_chirps):
        for n in range(8):
            data_radar[k, m, n] = adc_data[k, m, n]

# range FFT
range_win = choose_windows('Hamming', n_samples)  # add Hamming Windowing
doppler_win = choose_windows('Hamming', n_chirps)
range_profile = np.zeros((256, 64, 256)).astype(complex)
for k in range(256):
    for m in range(n_chirps):
        temp = data_radar[:, m, k] * range_win  # Hamming Windowing
        temp_fft = np.fft.fft(temp, N)  # Do N-point FFT for each chirp
        range_profile[:, m, k] = temp_fft
print(range_profile.shape)  # (range, chirps, Rx)

# doppler FFT
doppler_profile = np.zeros((256, 64, 256)).astype(complex)
for k in range(256):  # Rx
    for n in range(N):  # range
        temp = range_profile[n, :, k] * doppler_win
        temp_fft = np.fft.fftshift(np.fft.fft(temp, M))
        doppler_profile[n, :, k] = temp_fft
print(doppler_profile.shape)  # (range, doppler, Rx)

# angle FFT
angle_profile = np.zeros((256, 64, 256)).astype(complex)
for n in range(N):  # range
    for m in range(M):  # chirps
        temp = doppler_profile[n, m, :]
        temp_fft = np.fft.fftshift(np.fft.fft(temp, Q))
        angle_profile[n, m, :] = temp_fft
print(angle_profile.shape)  # (range, doppler, angle)

Acciwu, 2023-12-11:
> Thank you for your brilliant work first! And I am so sorry to bother you. I transform the raw ADC data with a (sample, chirps, Tx, Rx) structure to (sample, chirps, vritualRx), which is (256, 64, [...]

# Imports
import numpy as np
import matplotlib.pyplot as plt
# Read in frame data: ADC data
frame = np.load('./000028.npy')
frame = frame.reshape((256,64,8))
frame = np.transpose(frame, (1, 2,0))
# Manually cast to signed ints
frame.real = frame.real.astype(np.int16)
frame.imag = frame.imag.astype(np.int16)
print(f'Shape of frame: {frame.shape}')
# Meta data about the data
num_chirps = 64 # Number of chirps in the frame
num_samples = 256 # Number of ADC samples per chirp
num_tx = 2
num_rx = 4
num_vx = num_tx * num_rx # Number of virtual antennas
range_plot = np.fft.fft(frame, axis=2)
# Visualize Results
plt.imshow(np.abs(range_plot.sum(1)).T)
plt.ylabel('Range Bins')
plt.title('Interpreting a Single Frame - Range')
plt.show()

------------------------------------------------
range_doppler = np.fft.fft(range_plot, axis=0)
range_doppler = np.fft.fftshift(range_doppler, axes=0)
# Visualize Results
plt.imshow(np.log(np.abs(range_doppler).T).sum(1))
plt.xlabel('Doppler Bins')
plt.ylabel('Range Bins')
plt.title('Interpreting a Single Frame - Doppler')
plt.show()
range_doppler.shape
------------------------------------------------

num_angle_bins = 256
""" REMOVE """
padding = ((0,0), (0,num_angle_bins-range_doppler.shape[1]), (0,0))
range_azimuth = np.pad(range_doppler, padding, mode='constant')
range_azimuth = np.fft.fft(range_azimuth, axis=1)
range_azimuth = range_azimuth
range_azimuth.shape
------------------------------------------------

# Visualize Results
plt.imshow(np.log(np.abs(range_azimuth).sum(0).T))
plt.gca().invert_yaxis()  # Invert the y-axis
plt.gca().invert_xaxis()
plt.xlabel('Azimuth (Angle) Bins')
plt.ylabel('Range Bins')
plt.title('Interpreting a Single Frame - Azimuth')
plt.show()
[image]

cuixinyuan70, 2023-12-11:
Thanks for your help! :)

lijiunderstand, 2023-12-20:
> # Imports
> import numpy as np import matplotlib.pyplot as plt
>
> # Read in frame data: ADC data
> frame = np.load('./000028.npy') frame = frame.reshape((256,64,8)) frame = np.transpose(frame, (1, 2,0)) [...]

Hi, can you share the raw ADC data? Thank you

lijiunderstand, 2023-12-27:
> Thanks for your help! :)

Hi, can you share the ADC data? Thank you

Acciwu, 2023-12-29:
> Hi, can you share the raw ADC data? Thank you

I'm very sorry, but after careful comparison, this program is indeed not quite right

Acciwu, 2023-12-29:
> > Thanks for your help! :)
> 
> Hi, can you share the ADC data? Thank you

I only have department data, for example: https://drive.google.com/file/d/11FA6at2MzQ3g5G2T2scVBkHtLT1-9kgc/view?usp=sharing

lijiunderstand, 2023-12-29:
> department

Thank you very much for your help, can you share more with me as much as you can?

lijiunderstand, 2023-12-29:
> > > Thanks for your help! :)
> > 
> > 
> > Hi, can you share the ADC data? Thank you
> 
> I only have department data, for example: https://drive.google.com/file/d/11FA6at2MzQ3g5G2T2scVBkHtLT1-9kgc/view?usp=sharing

Thank you very much for your help, can you share more with me as much as you can?

Acciwu, 2023-12-30:
> > > > Thanks for your help! :)
> > > 
> > > 
> > > Hi, can you share the ADC data? Thank you
> > 
> > 
> > I only have department data, for example: https://drive.google.com/file/d/11FA6at2MzQ3g5G2T2scVBkHtLT1-9kgc/view?usp=sharing
> 
> Thank you very much for your help, can you share more with me as much as you can?

Of course, the ADC data is stored on the Google hard drive: https://drive.google.com/file/d/1R3dalVn9_x8GIH7lm3lZIUl1ftOzoJnp/view?usp=sharing.  If the RA image you processed matches the one provided by the author, can you share your code with everyone here?

lijiunderstand, 2023-12-31:
> Of course, the ADC data is stored on the Google hard drive: https://drive.google.com/file/d/1R3dalVn9_x8GIH7lm3lZIUl1ftOzoJnp/view?usp=sharing. If the RA image you processed matches the one provided [...]

Sure, I will. Really appreciate your help.

lijiunderstand, 2024-01-04:
> Of course, the ADC data is stored on the Google hard drive: https://drive.google.com/file/d/1R3dalVn9_x8GIH7lm3lZIUl1ftOzoJnp/view?usp=sharing. If the RA image you processed matches the one provided [...]

import numpy as np
import matplotlib.pyplot as plt
import math
import mkl_fft
import os

'''
"designed_frequency":			76.8 Hz,
"config_frequency":			    77 Hz,
"range_size":				    256,
"maximum_range":			    50 m,
"doppler_size":				    64,
"azimuth_size":				    256,
"range_resolution":			    0.1953125 m/bin,
"angular_resolution":			0.006135923 radian/bin,
"velocity_resolution":			0.41968030701528203 (m/s)/bin
'''

# def normalization(data):
#     norm = np.linalg.norm(data, axis=1, keepdims=True)
#     return data / norm

numSamplePerChirp = 256
numChirps = 64
range_res = 0.1953125

# read adc 

adc_path = "/mnt/RADDet_dataset/raw_adc/ADC/000080.npy"
rad_path ="/mnt/RADDet_dataset/train/train/RAD/all/000080.npy"
frame = np.load(adc_path)  

print(frame.shape) # shape: (256, 64, 4, 2)
frame = frame.reshape((256,64,8))  #(n_samples, n_chirps, nRv)
frame = frame.transpose((0,2,1)) #(n_samples, nRv, n_chirps)

print(frame.shape)

raw_adc = np.zeros(frame.shape,dtype = 'complex128', order='C')

w, d,h  = frame.shape

for c in range(d):
    temp = int((c%4)*2+c/4)
    raw_adc[:,c,:] = frame[:,temp,:]
frame = raw_adc

mean = frame.mean(0)
frame = frame - mean

range_fft = mkl_fft.fft(frame,numSamplePerChirp,axis=0)

dopplerFFTnum =64

RD_spectrums = mkl_fft.fft(range_fft,dopplerFFTnum,axis=2)
RD_spectrums = np.fft.fftshift(RD_spectrums, axes=2)  #(304,32,8)

RD_spectrums = np.flipud(np.fliplr(RD_spectrums)) 
rd = np.sum(np.abs(RD_spectrums), axis = 1) 

print('rd.shape', rd.shape)
plt.subplot(221)

plt.imshow(np.log10(rd))
print('RD_spectrums.shape', RD_spectrums.shape)

    
angle_fft_in = RD_spectrums   
angleFFTnum = 256 

angle_profile = mkl_fft.fft(angle_fft_in, angleFFTnum , axis =1)
angle_profile = np.fft.fftshift(angle_profile, axes =1)

# plot range-azimuth heatmap

ra = np.sum(np.abs(angle_profile), axis = 2)
print('ra.shape', ra.shape)

# normalization(angle_profile)

plt.subplot(222)
plt.imshow(np.log(ra))

RAD = RD_spectrums
print('RAD.shape', RAD.shape)

##################################################################

# read RAD
data =np.load(rad_path)
print(data.shape)  #(256, 256, 64)

plt.subplot(223)

#plot range doppler heatmap
RD = np.sum(np.abs(data), axis=1)
print('RD.shape',RD.shape) # (256, 64)
numSamplePerChirp = 256
numChirps = 64
range_res = 0.1953125
plt.imshow(np.log(RD))  

# plot range azimuth heatmap
RA_spectrums =np.sum(np.abs(data), axis=2)
print("RA_spectrums.shape",RA_spectrums.shape) 
plt.subplot(224)
plt.imshow(np.log(RA_spectrums))
plt.show()

Hi, here is the code for your reference.

lijiunderstand, 2024-01-04:
> Of course, the ADC data is stored on the Google hard drive: https://drive.google.com/file/d/1R3dalVn9_x8GIH7lm3lZIUl1ftOzoJnp/view?usp=sharing. If the RA image you processed matches the one provided [...]

import numpy as np
import matplotlib.pyplot as plt
import math
import mkl_fft
import os

'''
"designed_frequency":			76.8 Hz,
"config_frequency":			    77 Hz,
"range_size":				    256,
"maximum_range":			    50 m,
"doppler_size":				    64,
"azimuth_size":				    256,
"range_resolution":			    0.1953125 m/bin,
"angular_resolution":			0.006135923 radian/bin,
"velocity_resolution":			0.41968030701528203 (m/s)/bin
'''

# def normalization(data):
#     norm = np.linalg.norm(data, axis=1, keepdims=True)
#     return data / norm

numSamplePerChirp = 256
numChirps = 64
range_res = 0.1953125

# read adc 

adc_path = "/mnt/RADDet_dataset/raw_adc/ADC/000080.npy"
rad_path ="/mnt/RADDet_dataset/train/train/RAD/all/000080.npy"
frame = np.load(adc_path)  

print(frame.shape) # shape: (256, 64, 4, 2)
frame = frame.reshape((256,64,8))  #(n_samples, n_chirps, nRv)
frame = frame.transpose((0,2,1)) #(n_samples, nRv, n_chirps)

print(frame.shape)

raw_adc = np.zeros(frame.shape,dtype = 'complex128', order='C')

w, d,h  = frame.shape

for c in range(d):
    temp = int((c%4)*2+c/4)
    raw_adc[:,c,:] = frame[:,temp,:]
frame = raw_adc

mean = frame.mean(0)
frame = frame - mean

range_fft = mkl_fft.fft(frame,numSamplePerChirp,axis=0)

dopplerFFTnum =64

RD_spectrums = mkl_fft.fft(range_fft,dopplerFFTnum,axis=2)
RD_spectrums = np.fft.fftshift(RD_spectrums, axes=2)  #(304,32,8)

RD_spectrums = np.flipud(np.fliplr(RD_spectrums)) 
rd = np.sum(np.abs(RD_spectrums), axis = 1) 

print('rd.shape', rd.shape)
plt.subplot(221)

plt.imshow(np.log10(rd))
print('RD_spectrums.shape', RD_spectrums.shape)

    
angle_fft_in = RD_spectrums   
angleFFTnum = 256 

angle_profile = mkl_fft.fft(angle_fft_in, angleFFTnum , axis =1)
angle_profile = np.fft.fftshift(angle_profile, axes =1)

# plot range-azimuth heatmap

ra = np.sum(np.abs(angle_profile), axis = 2)
print('ra.shape', ra.shape)

# normalization(angle_profile)

plt.subplot(222)
plt.imshow(np.log(ra))

RAD = RD_spectrums
print('RAD.shape', RAD.shape)

##################################################################

# read RAD
data =np.load(rad_path)
print(data.shape)  #(256, 256, 64)

plt.subplot(223)

#plot range doppler heatmap
RD = np.sum(np.abs(data), axis=1)
print('RD.shape',RD.shape) # (256, 64)
numSamplePerChirp = 256
numChirps = 64
range_res = 0.1953125
plt.imshow(np.log(RD))  

# plot range azimuth heatmap
RA_spectrums =np.sum(np.abs(data), axis=2)
print("RA_spectrums.shape",RA_spectrums.shape) 
plt.subplot(224)
plt.imshow(np.log(RA_spectrums))
plt.show()

Hi, here is the code for your reference

lijiunderstand, 2024-01-04:
> Of course, the ADC data is stored on the Google hard drive: https://drive.google.com/file/d/1R3dalVn9_x8GIH7lm3lZIUl1ftOzoJnp/view?usp=sharing. If the RA image you processed matches the one provided [...]

import numpy as np
import matplotlib.pyplot as plt
import math
import mkl_fft
import os

'''
"designed_frequency":			76.8 Hz,
"config_frequency":			    77 Hz,
"range_size":				    256,
"maximum_range":			    50 m,
"doppler_size":				    64,
"azimuth_size":				    256,
"range_resolution":			    0.1953125 m/bin,
"angular_resolution":			0.006135923 radian/bin,
"velocity_resolution":			0.41968030701528203 (m/s)/bin
'''

numSamplePerChirp = 256
numChirps = 64
range_res = 0.1953125

adc_path = "/mnt/RADDet_dataset/raw_adc/ADC/000080.npy"
rad_path ="/mnt/RADDet_dataset/train/train/RAD/all/000080.npy"

# read adc
frame = np.load(adc_path)  

print(frame.shape) 
frame = frame.reshape((256,64,8))  
frame = frame.transpose((0,2,1)) 

print(frame.shape)

raw_adc = np.zeros(frame.shape,dtype = 'complex128', order='C')

w, d,h  = frame.shape

for c in range(d):
    temp = int((c%4)*2+c/4)
    raw_adc[:,c,:] = frame[:,temp,:]
frame = raw_adc

mean = frame.mean(0)
frame = frame - mean

range_fft = mkl_fft.fft(frame,numSamplePerChirp,axis=0)

dopplerFFTnum =64

RD_spectrums = mkl_fft.fft(range_fft,dopplerFFTnum,axis=2)
RD_spectrums = np.fft.fftshift(RD_spectrums, axes=2)  #(304,32,8)

RD_spectrums = np.flipud(np.fliplr(RD_spectrums)) 
rd = np.sum(np.abs(RD_spectrums), axis = 1) 

print('rd.shape', rd.shape)
plt.subplot(221)

plt.imshow(np.log10(rd))
print('RD_spectrums.shape', RD_spectrums.shape)

    
angle_fft_in = RD_spectrums   
angleFFTnum = 256 

angle_profile = mkl_fft.fft(angle_fft_in, angleFFTnum , axis =1)
angle_profile = np.fft.fftshift(angle_profile, axes =1)

ra = np.sum(np.abs(angle_profile), axis = 2)
print('ra.shape', ra.shape)

plt.subplot(222)
plt.imshow(np.log(ra))

RAD = RD_spectrums
print('RAD.shape', RAD.shape)

##################################################################

# read RAD
data =np.load(rad_path)
print(data.shape) 

plt.subplot(223)

RD = np.sum(np.abs(data), axis=1)
print('RD.shape',RD.shape) 
numSamplePerChirp = 256
numChirps = 64
range_res = 0.1953125
plt.imshow(np.log(RD))  

RA_spectrums =np.sum(np.abs(data), axis=2)
print("RA_spectrums.shape",RA_spectrums.shape) 
plt.subplot(224)
plt.imshow(np.log(RA_spectrums))
plt.show()

Hi, here is the code for your reference.

alenjames456, 2025-01-04:
> > > > > > Thanks for your help! :)
> > > > > 
> > > > > 
> > > > > Hi, can you share the ADC data? Thank you
> > > > 
> > > > 
> > > > I only have department data, for example: https://drive.google.com/file/d/11FA6at2MzQ3g5G2T2scVBkHtLT1-9kgc/view?usp=sharing
> > > 
> > > 
> > > Thank you very much for your help, can you share more with me as much as you can?
> > 
> > 
> > Of course, the ADC data is stored on the Google hard drive: https://drive.google.com/file/d/1R3dalVn9_x8GIH7lm3lZIUl1ftOzoJnp/view?usp=sharing. If the RA image you processed matches the one provided by the author, can you share your code with everyone here?
> 
> import numpy as np import matplotlib.pyplot as plt import math import mkl_fft import os
> 
> ''' "designed_frequency": 76.8 Hz, "config_frequency": 77 Hz, "range_size": 256, "maximum_range": 50 m, "doppler_size": 64, "azimuth_size": 256, "range_resolution": 0.1953125 m/bin, "angular_resolution": 0.006135923 radian/bin, "velocity_resolution": 0.41968030701528203 (m/s)/bin '''
> 
> numSamplePerChirp = 256 numChirps = 64 range_res = 0.1953125
> 
> adc_path = "/mnt/RADDet_dataset/raw_adc/ADC/000080.npy" rad_path ="/mnt/RADDet_dataset/train/train/RAD/all/000080.npy"
> 
> # read adc
> frame = np.load(adc_path)
> 
> print(frame.shape) frame = frame.reshape((256,64,8)) frame = frame.transpose((0,2,1))
> 
> print(frame.shape)
> 
> raw_adc = np.zeros(frame.shape,dtype = 'complex128', order='C')
> 
> w, d,h = frame.shape
> 
> for c in range(d): temp = int((c%4)*2+c/4) raw_adc[:,c,:] = frame[:,temp,:] frame = raw_adc
> 
> mean = frame.mean(0) frame = frame - mean
> 
> range_fft = mkl_fft.fft(frame,numSamplePerChirp,axis=0)
> 
> dopplerFFTnum =64
> 
> RD_spectrums = mkl_fft.fft(range_fft,dopplerFFTnum,axis=2) RD_spectrums = np.fft.fftshift(RD_spectrums, axes=2) #(304,32,8)
> 
> RD_spectrums = np.flipud(np.fliplr(RD_spectrums)) rd = np.sum(np.abs(RD_spectrums), axis = 1)
> 
> print('rd.shape', rd.shape) plt.subplot(221)
> 
> plt.imshow(np.log10(rd)) print('RD_spectrums.shape', RD_spectrums.shape)
> 
> angle_fft_in = RD_spectrums angleFFTnum = 256
> 
> angle_profile = mkl_fft.fft(angle_fft_in, angleFFTnum , axis =1) angle_profile = np.fft.fftshift(angle_profile, axes =1)
> 
> ra = np.sum(np.abs(angle_profile), axis = 2) print('ra.shape', ra.shape)
> 
> plt.subplot(222) plt.imshow(np.log(ra))
> 
> ##################################################################
> 
> # read RAD
> data =np.load(rad_path) print(data.shape)
> 
> plt.subplot(223)
> 
> RD = np.sum(np.abs(data), axis=1) print('RD.shape',RD.shape) numSamplePerChirp = 256 numChirps = 64 range_res = 0.1953125 plt.imshow(np.log(RD))
> 
> RA_spectrums =np.sum(np.abs(data), axis=2) print("RA_spectrums.shape",RA_spectrums.shape) plt.subplot(224) plt.imshow(np.log(RA_spectrums)) plt.show()
> 
> Hi, here is the code for your reference.

Could you please confirm if the provided code correctly converts raw radar data with dimensions (256, 64, 4, 2) into RAD data with dimensions (256, 256, 64)?

gincyvarghesec, 2025-04-24:
> import numpy as np import matplotlib.pyplot as plt import math import mkl_fft import os
>
> ''' "designed_frequency": 76.8 Hz, "config_frequency": 77 Hz, "range_size": 256, "maximum_range": 50 m, [...]

Hi,
   Can you please share RAD data, as the links for downloads are no more there.