ZhangAoCanada/RADDet #42: Request for pretrained weights

b4u365, 2023-10-25:
Hi Team,

Appreciate your great Work,

I am a Grad student and interested in radar research's, can you help me in getting the pretrained weights, currently the google drive link provided is not working.

This will be great help for my research.

Thanks and Regards,
Bharath.

Gabo181, 2023-11-19:
Push, would love to see this.

dongyu-du, 2023-11-22:
Yeah! The google drive link is still not available now. Could the author please check it?

YDMYy, 2023-11-24:
> Hi Team,
> 
> Appreciate your great Work,
> 
> I am a Grad student and interested in radar research's, can you help me in getting the pretrained weights, currently the google drive link provided is not working.
> 
> This will be great help for my research.
> 
> Thanks and Regards, Bharath.

This model can be trained on its own, but it seems that my accuracy dropped from 53 after running 1/5 of the data

dongyu-du, 2023-11-24:
Hi Bharath,Recently I am also working on training this model. Could you tell me your train configuration in config.json? And did you train it on GPU or CPU? If you use GPU, what’s the cuda and tensorflow version you use? Thank you!Best, Dongyu

YDMYy, 2023-11-24:
CUDA：11.8
Tensorflow:use tensorflow-gpu 2.1.0,Attempted 2.3, but it seems unsuccessful。

YDMYy, 2023-11-24:
CUDA：11.8
Tensorflow:use tensorflow-gpu 2.1.0,Attempted 2.3, but it seems unsuccessful，so i select 2.1.0

YDMYy, 2023-11-24:
For config.json,I only modified the dataset path in data to my own.

dongyu-du, 2023-11-25:
I think maybe the training is depended on the environment. What's your python version? Do you train the model using GPU? What type of your GPU?

YDMYy, 2023-11-26:
> I think maybe the training is depended on the environment. What's your python version? Do you train the model using GPU? What type of your GPU?

My Python version is 3.6 and I am using GPU, which is CUDA 11.8 and tensorflow gpu version 2.1.0. Here is a reminder that excessively high CUDA and tensorflow may not work, so it is best not to exceed this limit.

YDMYy, 2023-11-26:
> I think maybe the training is depended on the environment. What's your python version? Do you train the model using GPU? What type of your GPU?

I recommend you to download anaconda and create a virtual environment.because it's really convenient.

dongyu-du, 2023-11-26:
But I notice that the offical tensorflow version for cuda 11.8 is 2.12-2.14?(https://tensorflow.google.cn/install/source?hl=en). Can you run successfully using 2.1.0?
Thanks for advice, I use anaconda~

YDMYy, 2023-11-26:
> But I notice that the offical tensorflow version for cuda 11.8 is 2.12-2.14?(https://tensorflow.google.cn/install/source?hl=en). Can you run successfully using 2.1.0? Thanks for advice, I use anaconda~

it's ok.use pip tensorflow-gpu=2.1.0,if you run it on linux.About cuda 11.8,because my machine is 4090RTX,so I don't use the lower cuda.

dongyu-du, 2023-11-26:
Okay, Let me try~ what's your problem? My problem is that the model cannot converge when I use tensorflow 2.5.0, cuda 11.2 python 3.6, like this:
[image]

YDMYy, 2023-11-26:
> Okay, Let me try~ what's your problem? My problem is that the model cannot converge when I use tensorflow 2.5.0, cuda 11.2 python 3.6, like this: <img alt="image" width="1220" src="https://private-user-images.githubusercontent.com/58068933/285627859-534f292e-6744-4a6c-b9fc-a741fdfa4260.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDA5NzEwNjIsIm5iZiI6MTcwMDk3MDc2MiwicGF0aCI6Ii81ODA2ODkzMy8yODU2Mjc4NTktNTM0ZjI5MmUtNjc0NC00YTZjLWI5ZmMtYTc0MWZkZmE0MjYwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzExMjYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMTI2VDAzNTI0MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWZjYmI4MmZmZTU5Y2U2NTlkNmQzNDVjMTczODJiODI3YzAwZTMzYzFhM2JlNjdjNTg4OTMzY2MyMTVjNjAzZDkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Bw08lxRYzAvv1fhsSz2EAxESLpAoNxn7iZIMS4sUydw">

My question seems different from yours. My version conflicts between TensorFlow and some other Python libraries, while yours seems more like a data issue，You can try lowering the version and give it a try。

dongyu-du, 2023-11-26:
Sure, will do. We can discuss it if you would like. My email is dudy19@tsinghua.edu.cn

YDMYy, 2023-11-26:
> Sure, will do. We can discuss it if you would like. My email is [dudy19@tsinghua.edu.cn](mailto:dudy19@tsinghua.edu.cn)

I am very willing to stay in touch with you and engage in research in the field of radar.

YDMYy, 2023-11-26:
> > Sure, will do. We can discuss it if you would like. My email is [dudy19@tsinghua.edu.cn](mailto:dudy19@tsinghua.edu.cn)
> 
> I am very willing to stay in touch with you and engage in research in the field of radar.

你在清华上学嘛，我很菜的啦，哈哈，我的网易邮箱是katohui00@163.com.