ZhangAoCanada/RADDet #43: Radar Data Cube Size & own Radar Data

Gabo181, 2024-01-15:
Hi!

I'm currently trying to train a checkpoint with your onedrive train set - after that i want to use it to classify real-time radar data coming from an AWR2243BOOST, which has the radar data cube size 128, 12, 256 (Chirps, Virutal RX, Samples). 
I saw that you cumulate 3 frames to one and have the size 256, 256, 64.

Will i be able to run the detection on my radar frames or should i somehow also get my data into the same shape?

Thanks a lot!

ZhangAoCanada, 2024-01-19:
Hi, if my memory is right. The size of Chirps (128 in your case) can be configured within the TI radar itself. Normally, higher the number of chirps, higher the computational consumption. Make sure you have a good board/computer. btw, the radar cube (sized 256, 128, 12 in your case) is probably ADC data. You may have to perform FFT on it before feeding into the model.

All the data I had used for model training and testing are sized (256, 256, 64). So I don't know if the model can run on (256, 126, 64).

Gabo181, 2024-02-22:
Hi,

Your ADC Data is: 256, 64, 4,2
256 Sampels
64 Chirps
4 RX
2 TX?

Somewhere you mentioned, you collect 8 Chirps per Frame and add up 64 Frames. That wouwld mean your ADC Data is:
256 Samples
64 RX
8 Chirps, is that correct?