ZhangAoCanada/RADDet #52: Do train.py and train_cart.py share the totally same parameters?

huiwenXie, 2024-06-17:
Do train.py and train_cart.py share the same number of steps and the totally same parameters in 
`"TRAIN" :
		{
				"if_validate": true,
				"epochs" : 1000,
				"batch_size" : 3,
				"focal_loss_iou_threshold" : 0.3,
				"log_dir": "./logs/RadarResNet",
				"learningrate_init": 0.0001,
				"learningrate_end": 0.00001,
				"learningrate_decay": 0.96,
				"learningrate_decay_gap": 3000,
				"startup_steps": 1000,
				"learningrate_startup": 0.0001,
				"warmup_steps": 20000,
				"validate_start_steps": 2000,
				"validate_gap": 2000
		},
`?
Thanks a lot!

taj1290, 2024-08-08:
yes, as i tried the same so the results are ok.

huiwenXie, 2024-08-15:
> yes, as i tried the same so the results are ok.

Thanks a lot. Here

> yes, as i tried the same so the results are ok.

Thank a lot. I want to ask you another question. Can you tell me if the CONFIDENCE LOSS goes down for a while and then goes up in your results?
[image]
Here is the picture of my result.