ZhangAoCanada/RADDet #57: Missing ground truths for 000527-000532

FrzMtrsprt, 2025-01-17:
I noticed that the ground truth for 000527 to 000532 are missing from the OneDrive folder, their pickle files are neither in `/test/gt` nor in `/train/gt`. Would be nice if someone is willing to share a complete set of ground truth files.

sagar1899, 2025-03-05:
Hi,
can you share data including gt, npy and jpg folder

FrzMtrsprt, 2025-03-05:
> Hi, can you share data including gt, npy and jpg folder

Sorry I didn't download the image data.

gincyvarghesec, 2025-04-24:
@FrzMtrsprt ,
   Can you please share the gt and .npy folder , seems like they removed the link to download.

FrzMtrsprt, 2025-04-24:
> [@FrzMtrsprt](https://github.com/FrzMtrsprt) , Can you please share the gt and .npy folder , seems like they removed the link to download.

See #51.