ZhangAoCanada/RADDet #61: the shape of data

5p6, 2025-04-24:
thanks to author;s provide!
i download the raw ADC data,and then i check the shape of data,i found it is (256,64,4,2),it does not like the shape (256,64,256) as author say.

[image]
how to fix it or process it to train the model?

FrzMtrsprt, 2025-04-24:
If you want to transform them into RAD data cubes, you can refer to the code here.
https://github.com/jgiroux8/T_FFTRadNet/blob/main/RADDet/dataset/dataset.py

PH2021009, 2025-05-02:
@FrzMtrsprt , Can you please share the RAD data cubes and the corresponding images along with ground truths.

PH2021009, 2025-05-02:
Thanks @FrzMtrsprt  for responding.
How to evaluate its ground truth with actual scene. Any ideas on this ?

FrzMtrsprt, 2025-05-02:
> Thanks [@FrzMtrsprt](https://github.com/FrzMtrsprt) for responding. How to evaluate its ground truth with actual scene. Any ideas on this ?

You can refer to the code here.
https://github.com/ZhangAoCanada/RADDet/blob/main/metrics/mAP.py